jwise/x1-expander #1: pcb fixes

bani6809, 2024-05-31:
**rj45 connector looks potentially incorrect, needs review. pinout does not match datasheet**

C47 orientation can be improved - rotate 90'
USB esd protcetion are potentially stubs
GND via near Net-U3-SW pad 1 is useless
increase margin near net-u3-sw-1 pad
concern that the 5v above the camo_dp may have induced currents from the switcher
xtal for lan9514 on the back, may have excess parasitic capacitance and xtalk. keep lengths short and avoid layer changes
adbus0, adbus1 acute angles
potentially redefine camo_dp and ftdi_dp so the differential pairs do not have to cross (choice of ports is arbitrary)
use internal polygon plane for vcore on ftdi_dp
vdd18core can potentially use polygon plane in 2nd layer to clear up top layer
save a whole $1.43 on the order by putting silkscreen "JLCJLCJLCJLC" under one of the gpio headers
add more vias for noise suppression. there are some but could be more
consider adding diodes for reverse polarity protection, add idiot proofing to the board. users will see 5v and 3v and try to feed it power

bani6809, 2024-05-31:
i verified the easyeda footprint matches the datasheet for the following rj45 connectors:

https://www.lcsc.com/product-detail/Ethernet-Connectors-Modular-Connectors-RJ45-RJ11_HANRUN-Zhongshan-HanRun-Elec-HR911103A_C375844.html

https://www.lcsc.com/product-detail/Ethernet-Connectors-Modular-Connectors-RJ45-RJ11_HCTL-HC911102A-904-A_C5341216.html

https://www.lcsc.com/product-detail/Ethernet-Connectors-Modular-Connectors-RJ45-RJ11_HCTL-HC913129AE-907-A_C5341218.html

jwise, 2024-05-31:
LAN1 replaced with C375844, 7b0b5fa

jwise, 2024-05-31:
C41 rotated, c550f1c

jwise, 2024-05-31:
Agree that USB ESD are stubs but I am not super duper worried in this case.  The longest route is 9.5mm; with a propagation velocity of 60%, that's 52ps.  On a 480MHz signal, that's 2083ps, so we are well well well well *well* below lambda/4.  I will waive that one.

Useless via removed.

Increased clearance on net between U3 and L1 to 0.3mm.

Modified 5V routing to go away from `CAMO_DN`.  `CAMO_DP` still has a relatively tight ground route around there but hopefully the return current should go on the inner planes rather than through there.  23119ad

jwise, 2024-05-31:
ADBUS0 and ADBUS1 routing repaired in e314819.

jwise, 2024-05-31:
Vcore on FTDI moved to In2.Cu in 2591d1b.

jwise, 2024-05-31:
Vdd18core moved to inner copper layer.  But to be honest all of that routing is a clusterfuck that I am not proud of.  The less said about it the better.  If I were to redo it I would have kept the 0402s and I would move FB1 north so that I could more easily route the differential USB traces around it.  cb720ee

Waiving the camo and ftdi cross.  Probably a good idea but more effort now than it is worth to reroute.  

My dignity is worth $1.43.  Waived.

Stapled ground planes with 5mm via grid where design rules permitted.  9cd169a

Waiving reverse polarity protection / idiotproofing.  Fuck around and find out.

I agree with the xtal thing, the layout guidelines did say to place it far away from other things but I'm not sure why I put it on that side.  I'll get to that one tomorrow.

bani6809, 2024-05-31:
concern that the single 5v via + trace may not provide enough current.
extend 5v plane polygon to the GPIO 5V pins and bottom layer traces can probably be removed.

bani6809, 2024-05-31:
i dont care if users blow up the board but we should think about ways they could short out the expander board or feed power back into the AP and blow it out. basically, direct paths from any of the external connectors back to the AP.

uart is unprotected and that goes right into the rockchip does it not.

jwise, 2024-06-01:
UART protection in b5f2906b44e1ea9977de329c13b2cd18d99430af

gpio 5v fattened in 2402ba38cf76c6adddf3df264edbf50903e8b924

jwise, 2024-06-01:
xtal moved up to the top in d1a6997

closed 2024-06-01 by jwise