jwise/x1-expander #13: USB ports interfere with each other

jwise, 2024-06-11:


bani6809, 2024-06-13:
USB official spec defines the allowed type A plug overmold dimensions:

[image]

16mm x 8mm would seem to be a good minimum

jwise, 2024-06-13:
I am thinking that a vertical USB socket would be a good approach to avoid having to change the dimensions of the board.

bani6809, 2024-06-13:
something like this one then

https://www.lcsc.com/product-detail/USB-Connectors_SHOU-HAN-AF-14-0-CC-T_C456020.html

jwise, 2024-06-13:
yeah, that looks like a good option

bani6809, 2024-06-14:
this may be easier to route:
https://www.lcsc.com/product-detail/USB-Connectors_XUNPU-USB-226-BRW_C720524.html

jwise, 2024-06-14:
hmm though with the EEPROM we can configure the LAN9514 to swap P and N for us, but I will look at that one too

closed 2024-06-26 by jwise