jwise/x1-expander #36: formalize+document expander board eeprom

bani6809, 2024-07-25:
formalize+document expander board eeprom

e.g. d6-clk/d7-dat with i2c eeprom of minimum X byte capacity, and the following structured content.

zombu2, 2025-01-20:
just out of curiosity what does this eeprom do ?

jwise, 2025-01-21:
I need to document both onboard and module EEPROMs, but they are:
* for mainboard: lan9514 config data
* model number identifier
* serial number identifier
* hash and signature of model + serial number to verify authentic modules (unsigned EEPROMs will work perfectly fine without, they will just display a warning on boot in official X1Plus firmware that sad-face please buy the real thing to support the project... it's open source so patch it out if you must)

zombu2, 2025-01-21:
oh aright well thank you for the info i was just curious