jwise/x1-expander #76: Did I need to flash extra firmware on the expander board if I make the board on my own

LE1ED4523, 2024-12-08:
I see some sign eeprom thing and bin file there but I'm not familiar with programming.

jwise, 2024-12-08:
You will want to flash the descriptors for the LAN9514.  It should work fine without it but it is really better to mark the board as for which version it is.  Speaking of versions, you probably want to wait until the final version before you fab out any of your own.  If you fab out a FTDI-based board right now, it will not be supported in the future (and it will support only two ports); if you fab out a RP2040-based board right now, the firmware is not ready for it yet.

The sign-eeprom thing creates a signed serial number EEPROM image.  The board will work just fine without a signed EEPROM and everything will be functional, but it will complain once on each boot saying that buying an authentic Expander helps support the project :)

My suspicion is that it probably will be similarly expensive to make 5 as it would be to buy 5 from Crowd Supply.  Above a volume of 5 or 10, it might make sense, but then you take on all the yield risk yourself, too, and you're going to have to work with suppliers directly, which sounds like it is not fun.  If you want a lot of them, send me some e-mail through the Crowd Supply page and we'll work out if we can get you a discount!

LE1ED4523, 2024-12-08:
Thanks for the quick reply,I'll reconsider get it from website.Wish your project all the best :)

closed 2024-12-08 by LE1ED4523